sheldonucr/ucr-eecs168-lab #1: cdesigner& Issue

lgarc025, 2017-01-11:
I can log in to bender, but when I use "cdesigner&" command to test it I get the following error,

[1] 26628
bender /home/eemaj/lugarcia $ Error: DISPLAY environment variable is not defined.

[image]

Cwcook2, 2017-01-11:
Hi Luis,

We can discuss the issue during lab as attendance is required.

However, I suspect you missed a step in the lab procedure and likely do not have your X-client configured correctly. If you resolve your issue, please update this thread incase other student encounter the same problem.

lgarc025, 2017-01-11:
If on a Mac, make sure you have XQuartz on your machine!

zergzerg, 2017-01-11:
Hi everyone, 

If you are using your own laptop to login the bender and blocked by your firewall, you can try the vpn which can be found at http://systems.engr.ucr.edu/vpn/vpn.html .

closed 2018-03-02 by Cwcook2