sheldonucr/ucr-eecs168-lab #5: Testing lab 2 problems

AlexArdelean, 2017-01-31:
We are having trouble running the testbench. We can't get the right AIN and BIN vpulse wave forms. We set the AIN and VIN vpulse's to the same properties  as lab 1 except changed the specified pulse width to what was said on the lab. Are these the right settings? This is what we get. 

[image]

[image]

We think it's from setting up the dc analyses from the SAE. We have 2 vpulse's AIN and BIN, but when setting up the dc sweep on the Transient Simulation Setup we don't know what settings to use since we have 2 vpulses and the setup asks you to choose only 1. On this part. 

[image]

Cwcook2, 2017-01-31:
Firstly, you only need to set up a transient simulation. We don't care about the DC simulation for this lab.

Secondly, you must modify the period of the signal, not somply the pulse width. Please take the time to read up on a signal's period, pulse width, rise/fall times, etc. as these are things you should be very familiar with.

AlexArdelean, 2017-01-31:
Thanks we figured out what we were doing wrong, our pulse width was wider than the period, we are having another small issue tho. We can't figure out how to add the capacitance load (0.05p F) at the output to remove noise of OUT. We tried adding a cap from analoglib with the correct pF right after our output but we don't receive anything from the other end of the capacitor. We also tried attaching the capacitor backwards.  

[image]

Cwcook2, 2017-01-31:
Try to remember from the lecture review what I said about capacitors. They only allow AC signals to pass.. What we want, is to have the noise ( typically an AC signal ) pass to VSS while we get the digital signal at the ouput, ie, you need to measure the output before the capacitor.