sheldonucr/ucr-eecs168-lab #6: Lab 2 - NAND Testing

ykamo001, 2017-01-31:
[image]
[image]

I am not able to get any output when running the simulation with the capacitance loaf of 0.05p F, but if I were to remove the capacitor all together, I am able to get an output with some noise.
[image]

I am not using the proper element to filter correctly, or is there something wrong with my schematic?

Cwcook2, 2017-01-31:
Try to remember from the lecture review what I said about capacitors. They only allow AC signals to pass. So we should expect to see exactly what you are getting when you place the output pin after the capacitor. What we want, is to have the noise ( typically an AC signal ) pass to VSS while we get the digital signal at the ouput, ie, you need to measure the output before the capacitor.

ykamo001, 2017-01-31:
That fixed it, and makes sense. Thank you!

closed 2017-01-31 by ykamo001