sheldonucr/ucr-eecs168-lab #8: Waveform is not executing

ldiaz011, 2017-02-03:
[image]

After running the transient analysis, I pass the tests but the waveform does not appear. It says I am missing a library or something? I have no idea what the issue is

zergzerg, 2017-02-03:
Can you check in the SAE simulation model files set up (Setup -> Models Files), did you change the BJT to TT_12?

ldiaz011, 2017-02-03:
Yes I did. I made sure of that already

Cwcook2, 2017-02-03:
The error looks like a library/dependency issue. Is there another, perhaps more detailed, error report in another window?

Cwcook2, 2017-02-03:
Actually, Can you take a screen capture of your model files setup?

ldiaz011, 2017-02-03:
[image]
Here it is

Cwcook2, 2017-02-04:
If you're able, you can come to our lab (WCH361) right now and we can take a look at. Otherwise, we can help you during lab time as it is due at 8:00.