sheldonucr/ucr-eecs168-lab #14: AVSS not matching with ground

magui028, 2017-02-16:
Hi,

I keep getting an error with LVS that I have a shorted net on my layout with gnd! and AVSS. 

Also the same with AVDD and vdd!

I'm a little lost on how to fix this

[image]
[image]

Cwcook2, 2017-02-16:
We would need to see your layout and schematic to be able to help. Chances are, the problem is exactly how you described it, you are shorting a net in your layout.

magui028, 2017-02-17:
Okay, should I post it here? or when can I come in?

Cwcook2, 2017-02-17:
You can post them here, email them to me, or email me to schedule an appointment.

magui028, 2017-02-17:
I figured out my error. I had some general oversight and used pmos3ts/nmost3ts instead of the 4ts so the grounds didn't match

closed 2017-02-17 by magui028