sheldonucr/ucr-eecs168-lab #19: Lab 3 Issue

donei003, 2017-02-22:
Hey I'm having issues with the lvs and can't seem to locate the problem. Could you help me with this?
[image]

zergzerg, 2017-02-22:
That means the lvs cannot locate the corresponding CMOS in the layout. Try to find out the corresponding CMOS in layout and check its connections. 

BTW, I find some errors in schematic and they are marked. Some input port is used for output and there is one cross section which is not allowed in the schematic. 

If you are still confusing, I can help you after class or during tomorrow's lab.

donei003, 2017-02-22:
So I'm still getting some LVS errors but I'm not sure why. I think the program thinks CIN and B are connected but they shouldn't be and upon review they're not. Could you take a look?

[image]

Cwcook2, 2017-02-23:
You can come to our office tomorrow if you would like to discuss this further. I can't see any errors really but it is somewhat hard to see. There is one section where it looks like your poly for Cin and B may be touching ( in the top right just below the parallel transistors.

donei003, 2017-02-23:
That sounds good to me, what time would you like to meet and is it the TA room too?

Cwcook2, 2017-02-23:
You can come to our lab (WCH 361) any time before 5:00pm today