sheldonucr/ucr-eecs168-lab #20: Waveform output issue

ColdNightHour, 2017-02-23:
I am currently trying to output the proper waveforms but keep getting jumbled stuff for my output. Do we need capacitative loads to filter any noise? If so, how much capacitance. (P.S. I already tried .05 pF loads at each output, and it still looks like crap)

Cwcook2, 2017-02-23:
This depends on how bad it looks. You can post a screenshot or email a screenshot to me and I can tell you if this is a parasitic/noise issue or if there is a more serious problem.

ColdNightHour, 2017-02-23:
[image]
It's also very possible I just didn't do it right

ColdNightHour, 2017-02-23:
This is with a capacitative load that's near to 0 (.0000025 pF, which I know is  outrageous but I was desperate.)

zergzerg, 2017-02-23:
Hi, I don't think this is noise, and capacitor cannot work for this case. You should check your design to find out the error.  If you need more help, you can make an appointment with me or Chase.

ColdNightHour, 2017-02-23:
Alright. I thought as much. Can I make an appointment for today, assuming I don't fix the error?

zergzerg, 2017-02-23:
OK, you can come to WCH 361 from 2 to 5.

ColdNightHour, 2017-02-23:
Fixed it. VSRC was connected to VSS instead of VDD.