sheldonucr/ucr-eecs168-lab #22: LVS Error for One Bit Adder: text_short

ldiaz011, 2017-02-28:
I have an error that I have no idea how to fix. It passes DRC but I get one error when I run LVS:
I put the screen shot of my error and layout 
[image]
[image]

Cwcook2, 2017-02-28:
I think you mixed up your p-implant and n-implant areas ( green and red boxes) unless you put your nmos on top and nmos on bottom. But it is hard to tell from the photo

Cwcook2, 2017-02-28:
The error you are getting is an extraction error also, not an LVS error. So maybe you did not name your nets correctly.

ldiaz011, 2017-02-28:
I switched the p and n implant and now I got this error:
[image]

Cwcook2, 2017-02-28:
unmatched nets mean that wires in your layout are not connected the same as your circuit. unmatched devices mean your have an extra transistor or are missing a transistor in your layout