sheldonucr/ucr-eecs168-lab #25: Lab 4 Verilog Issue

donei003, 2017-03-14:
I'm having issues getting my code to pass the test bench. I thought we went through all my code and validated it in lab but I guess there are some errors still. Could you take a look at my code? I've attached the three files
[Lab.zip](https://github.com/sheldonucr/ucr-eecs168-lab/files/842773/Lab.zip)

nmlapre, 2017-03-14:
Just wanted to add my name here - my code is very similar, and I'm having trouble passing the test cases as well.

ady002, 2017-03-14:
Same with me.

zergzerg, 2017-03-14:
Hi,
I checked your code and find some mistake in the control file. In the DONE state, you should set result_val to 1. 
BTW, make sure you use the correct mux_sel  for A and B.

donei003, 2017-03-14:
Could you be more specific? I added the part the sets result_val to 1 but I am still failing the test bench

Cwcook2, 2017-03-15:
Dylan,

Make sure the signals driving your muxes are correct. For example, a mux
may select in1 if sel=0 and in2 if sel=1. But you may be switching it
around in your control unit. So make sure the mux sel is selecting right
inputs.

-Chase

donei003, 2017-03-15:
Yup, had some connections flipped (ex: I had B_mux_sel = A_reg when 1'b0, and initiliazed B_mux_sel to 1'b0 so it was getting A_reg instead of operands). All good now, thanks for the explaination!