sheldonucr/ucr-eecs168-lab #30: Cannot open recent designs

brandonlee0405, 2018-01-31:
I cannot open any of the schematic/designs I have created that are in mylibrary even though the file is in my eecs168 folder and I opened cdesigner in the eecs168 folder. I cannot even access mylibrary on cdesigner

psythefirst, 2018-01-31:
This does not sound a typical problem. Can you show me this problem before or after today's lecture?

brandonlee0405, 2018-01-31:
Okay, I will show you prior to class

psythefirst, 2018-01-31:
This problem is caused by the file lib.defs.
After we created our own library 'mylibrary' in lab1, a new line 
`DEFINE mylibrary ./mylibrary`
is added to this file.
If you somehow have modified this file you will lose access to your library in cdesigner, try adding this line back to lib.defs to solve this problem.

closed 2018-01-31 by psythefirst