sheldonucr/ucr-eecs168-lab #32: Problem with SAE-Lab3 Full adder

ghazalmazaheri, 2018-02-08:
Hi,
When I set my input pulses in SAE, I faced the following error:
Expression 'v(/B)' did not evaluate: No results found

[image]

psythefirst, 2018-02-08:
The most possible reason is that your simulation did not run through. Can you try running it again and post the console output?

ghazalmazaheri, 2018-02-09:
Thanks for your answer. Here it is:

[image]
[image]
[image]

psythefirst, 2018-02-09:
I think you might have not properly set up the model file and the section because hspice is complaining that no models are found for pmos/nmos. Please refer to fig.21 in lab1 to double check whether they are good.