sheldonucr/ucr-eecs168-lab #33: Hw 2 solutions

uesco001, 2018-02-13:
For #4 a) I don't understand why the best case for pmos is 3/4 and 4/3 for nmos. I thought the best case was the least amount of mos you have to go through. So shouldn't that make it 1 for pmos and 2 for nmos?

[image]

psythefirst, 2018-02-13:
In the best case, all transistors of one kind are open. In this example, pmos part is one parallel with three and nmos part is 3 in parallel serial with another one. So the result is 3/4 PMOS and 4/3 NMOS.

uesco001, 2018-02-13:
I think an easy way to do it is to pretend they are resistor values that you don't know and try finding the total resistance. So for example in this probelm you have transistor a b c in series and d in parallel. You can say you have (Ra+Rb+Rc)||Rd so then you expand on this and just plug in 1 after you finished extracting it.

psythefirst, 2018-02-13:
Yes, this is a good way to do it. Just treat all nmos/pmos as Rn/Rp, and then get the equivalent resistance from vdd/gnd to output port.

IgneousFalcon, 2018-02-13:
I understand now. Thank you

tgadi001, 2018-02-14:
Hey we need to submit the topic. You want to do Solar Cells?

Sincerely yours,
Tahir Gadimov

closed 2018-03-02 by psythefirst