sheldonucr/ucr-eecs168-lab #34: Lab3 1BitFullAdder Layout Issue

jreyes07, 2018-02-17:
I've passed the DRC check for my system.  My issue occurs when I run the LVS check.  On the Console Window it shows that the LVS results are clean.  However, on a different window it shows that there are unmatched nets, etc.  Which one should I listen to and why are they contradicting each other?

psythefirst, 2018-02-17:
This is an interesting issue. I would trust the debugger more. Could you try to run LVS once again?

jreyes07, 2018-02-17:
I've tried it several times but I get the same thing. The Console window and *.LAYOUT_ERRORS window show clean.  The Run Summary window and debugger still show errors.

psythefirst, 2018-02-17:
Can you post a screenshot of the errors? I think you can try post simulation as check whether the logic is correct. If yes, this should be a small issue.

jreyes07, 2018-02-19:
[image]

Errors from my current 1 Bit Adder Layout

psythefirst, 2018-02-19:
They seem real problems for me. I would carefully go through and try to correct each of them. If you need help with debugging you can post more screenshots here or come to our office hour.

jreyes07, 2018-02-21:
So all of my PMOS' have matches but it's showing that there are unmatched NMOS'.  Does it matter which input is closer to ground when it comes to a NAND gate NMOS?

psythefirst, 2018-02-21:
Yes, it does matter because it results in different connections.

jreyes07, 2018-02-21:
Debugged all my issues. Thank you for all the help and advice!