sheldonucr/ucr-eecs168-lab #44: LVS problem

ralme002, 2019-01-28:
I am passing DRC I have rebuild the lab thinking I did something wrong but I am still getting same issues and can't figure out why I am having these errors.
[image]
[image]
[image]
[image]
[image]

jborj006, 2019-01-28:
Could it be that the Polysilicon (PO) is over .18?

ralme002, 2019-01-28:
I tried changing over and under .18 and still It didn't work. I think its something not matching between the Layout and the schematic. Feels like its something simple but I just don't know what it is.

hzhouplus, 2019-01-28:
Can you come to WCH 361 before 5:00pm today to show this issue?

ralme002, 2019-01-28:
Sorry I will be at work at that time. Is it possible tomorrow? I can meet anytime from 9am to 2pm.

hzhouplus, 2019-01-28:
Tomorrow is OK for us.

ralme002, 2019-01-28:
ok thank you

jborj006, 2019-01-28:
I think you didn’t make that large PO into a path. I don’t see the line running through the middle.

ralme002, 2019-01-28:
Good eye! I just fixed it. It was a rectangle not a path. I edited on my first post a new pic of what it looks like now.  However the problem is still there

ralme002, 2019-01-28:
[image]

closed 2019-02-10 by ralme002