sheldonucr/ucr-eecs168-lab #46: NAND gate testbench

codyiskhuu, 2019-01-29:
When doing the simulation for the testbench, for both A and B input pulse voltages, do we have to change the periods to get different outputs to test the NAND gate? Also instead of ground, what should I put for VSS?  A bit stuck trying to find the right timings to get all NAND gate results. 

[image]

psythefirst, 2019-01-29:
Yes, you need to change the period and pulse width to replicate the waveform we gave in the lab manual. Ground is good for VSS.

psythefirst, 2019-01-29:
More precisely, you want the period and pulse width of one vpulse double of the other vpulse.