sheldonucr/ucr-eecs168-lab #47: Lab 2 Part 8 Question

Tsukuyomiiii, 2019-01-29:
I after doing the transient simulation I got this result: 
[image]
I'm not sure if it is the correct output but the relationship between a,b and the output seems to be right. 

Also, where can I find the NAND gate layout? It says it's on the lecture notes but I couldn't find it. I may have overlooked it, though.

Thank you!

anniedu97, 2019-01-29:
Theres no direct NAND layout in the notes, but you can find the NAND stick diagram on the Chapter 2-4 Lecture notes.

psythefirst, 2019-01-29:
Regarding your simulation, please refer to #46

closed 2019-02-01 by psythefirst