sheldonucr/ucr-eecs168-lab #50: NAND_TESTBENCH

jborj006, 2019-01-29:
[image]
[image]
[image]
[image]
[image]

I have no errors when running my test-bench but it does not look like the one in the lab manual. Any hints on what I may be missing?

exoticmoose, 2019-01-29:
I believe you should set pulsewidth = period/2 for the VPULSE sources. Your current setup essentially just has two constant voltage sources which "drop out" down to zero momentarily at the end of their periods.

jborj006, 2019-01-29:
Gotcha thank you

closed 2019-01-29 by jborj006