sheldonucr/ucr-eecs168-lab #52: regarding running the DRC

cybertron97, 2019-02-01:
[image]

I have been getting this error , everytim i run the DRC

cybertron97, 2019-02-01:
I killed the process, and made it to run again

psythefirst, 2019-02-01:
This is a common problem this year. I am reopening the issue for other people who might run into this.
The solution is first finding out the PID of icv_vue and then killing the process.
To find out the PID, you can type `ps -u 'your ENGR ID'` in the terminal, or use `top` to list all running processes and find it.
To kill the process, just `kill PID_NUMBER`, for example, `kill 194` if 194 is the PID of icv_vue

cybertron97, 2019-02-01:
I guess even if you just write ps -u, it opens up only your user processes, also , it happens when you have left your mac on for a while on ucr wifi and a broken pipe occurs which creates this process and hence when you try to re-run it gives that error