sheldonucr/ucr-eecs168-lab #58: No voltage shown in custom desing consol Lab3 Full adder

jborj006, 2019-02-07:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
It is showing zero when I run my test bench. I can not figure it out how to fix it. I have posted pictures for your reference. 

Thank you

psythefirst, 2019-02-07:
Check your vpulse properties first

jborj006, 2019-02-07:
Vpulse: V2 = 1.2 V and tr =10ps. Right? That’s what I have.

psythefirst, 2019-02-07:
These two are good. Further check the delay time, fall time, pulse width and period. I see your 3 vpulses output 1.2V at the beginning but quickly dropping to 0, so it should be something wrong with their settings.

jborj006, 2019-02-07:
[image]
[image]

I tried changing vpulse. It made a slight change.

psythefirst, 2019-02-07:
Did you see your period is 4s?

jborj006, 2019-02-07:
[image]

It is looking better now. My problem is not with the COUT and SUM

jborj006, 2019-02-07:
[image]
Fixed it! There was a transistor that was not connected to the proper output. Thank you Peng:+1:

closed 2019-02-07 by jborj006