sheldonucr/ucr-eecs168-lab #66: Connecting layout M1 to pins without bus

danialesaid, 2019-02-20:
Hello,

Sorry to bother you.

I am having one error with lab3.

I created the schematic and layout for the onebit adder.
I am just having some trouble connecting to pins in the onebit adder layout.

I have M1 Pins declared using the M1PIN layer:
[image]

I then have M1 labels declared using the text for M1 layers:
[image]

The M1 layer text labels are not attached or expanded. 
In my experience, the M1 layer text labels are not mapped to the pin.

Is there anyway to map the wires to the pins in the layout editor without making a bus?

Thank you

psythefirst, 2019-02-20:
You should use M1Pin for text labels, like what you did in the first picture. I do not understand what you are trying to do in the second picture, because texts using M1drw layer make no sense to the layout. I do not understand your question of 'map the wires to the pins in the layout editor without making a bus' either. Can you explain more about it?

danialesaid, 2019-02-20:
Thanks for a prompt reply. 
In the schematic, I specified only one pin and used wire names to map different wires to the pin without a bus.

I thought that it worked similarly in the layout where there is only one pin and different wires may be connected to that pin using labels instead of a bus.

Thank you

psythefirst, 2019-02-20:
I see. You have to connect all wires together as the layout is for real manufacturing. Check #13