sheldonucr/ucr-eecs168-lab #70: Lab3 convergence aid

danialesaid, 2019-02-25:
Hello,

I hope this finds you well.

I am having some trouble understanding the convergence aid for the one bit adder and the four bit adders. When I run the simulation for the 4 bit adder without the convergence aid I get the following:

[image]

However with convergence aid I get what I believe is the same simulation result:
[image]

The same is true for the one bit adder.

With convergence aid:

[image]

Without:

[image]

Both are run with parasitic extraction.

Thank you

psythefirst, 2019-02-25:
Hi Daniel, we don't need convergence aid for the adder simulations.

danialesaid, 2019-02-25:
Thanks for a prompt reply. 

That sounds good. 

Is there anyway that I could stop by the lab today to get checked off?

Thank you

hzhouplus, 2019-02-25:
Can you come at 3:30PM?

danialesaid, 2019-02-25:
That sounds perfect.
Thanks for making the time.

See you then.