sheldonucr/ucr-eecs168-lab #71: Lab #3 LVS Error (text_net:text_short)

akiyowind, 2019-02-26:
I'm receiving an extraction error for my 1bit adder. I passed the DRC, but I'm not sure how I should go about debugging this error. Attached are pictures of my LVS output as well as my layout.

I marked the inputs & outputs (used M1Pin for the text labels) in the layout, as I know that it is hard to see.

[image]

[image]

psythefirst, 2019-02-26:
It means all the 7 networks are shorted. Please check the connection.

akiyowind, 2019-02-26:
I went to go grab food, and when I came back, I've been unable to run the LVS due to this strange error. Any idea on how to fix this?

[image]

psythefirst, 2019-02-26:
Please refer to #52

akiyowind, 2019-02-27:
Hi, so I've changed my layout (I realized some of my inputs were not connected properly), and VDD is no longer shorted. However, the rest of them still are. I have trouble understanding how A is shorted in the layout. Do you think I could come to OH tomorrow to get some assistance?

Changed layout:

[image]

psythefirst, 2019-02-27:
There seems to be a long M2 wire overlapped with an M1 wire shorting with
other M2.

akiyowind, 2019-02-28:
Hi,

I would like to thank you for your help earlier. I managed to debug almost all of my errors. However, I am having trouble understanding why my Cout  in the layout is not matching with the schematic. Do you think you could help me with this issue?

[image]

[image]

psythefirst, 2019-02-28:
I think it is because you have Cout' and Cout all named Cout in the schematic

akiyowind, 2019-02-28:
I am a buffoon. Thank you.

akiyowind, 2019-02-28:
Debugged all of my errors. Thanks again!