sheldonucr/ucr-eecs168-lab #75: Waveform Issue

Y0ungMun, 2019-03-01:
My waveforms for the 1-bit adder are identical for all of my inputs. My outputs are also the same.
[image]
I understand I am very far behind and will probably lose points for being so far behind, but dealing with PuTTY for these past few weeks has been rough with it freezing on me, grinding to a halt, and disconnecting randomly, and thus losing a lot of progress. At this point, I would much rather have the assignment done and submit something broken, but finished than nothing at all.

psythefirst, 2019-03-01:
Please refer to #46

Y0ungMun, 2019-03-01:
My waveform looks a little better, but now I'm getting a lot of error, apparently caused by floating instance pins. Any way to remedy this?
[image]

Y0ungMun, 2019-03-01:
Never mind, I think I got it.