sheldonucr/ucr-eecs168-lab #76: [LAB03] Stick Diagram Clarification

akiyowind, 2019-03-01:
Hi, 

I noticed that one of the deliverables of the Lab03 is a stick diagram of the 1-bit full adder. Did you want a stick diagram of specifically the layout we implemented, or will any stick diagram do?

Thanks,

Adrian

psythefirst, 2019-03-01:
Of course the one matching your layout design, if you have one. This is required in order to help you design, so you don’t need to work out a new one if you don’t have.

akiyowind, 2019-03-01:
I've completed the rest of the lab. I just skipped the stick diagram because I wasn't sure what was required. Just to make sure, since I have the layout completed already, I don't need to submit a stick diagram in my report?

psythefirst, 2019-03-01:
no

akiyowind, 2019-03-01:
Got it. Thank you!