sheldonucr/ucr-eecs168-lab #77: Lab 4 correction

geoffreynguyen011, 2019-03-07:
Is there a way for us to know if we did lab 4 correctly, so we can demo it properly next Wednesday? I want to make sure I don't waste time doing the lab all over again if it's wrong.

hzhouplus, 2019-03-07:
You just need to finish all the parts we mentioned for check off and demo it next Wednesday.

geoffreynguyen011, 2019-03-08:
I tried demoing this past Wednesday, but my compiler reports were not correct, according to my TA. Would that mean that even if my reports are not correct when I demo again on Wednesday, I will still get full credit for the lab?

hzhouplus, 2019-03-08:
You may need to try another time, most students get reference as 30.

geoffreynguyen011, 2019-03-08:
I tried another time, this time in a different folder. However, I don't want to repeat the same mistakes that I did last time. Would it be ok if I post my reports here, and you can tell me if they are correct or not?

hzhouplus, 2019-03-08:
Since I didn't see all the steps for you to do lab4, I cannot tell what mistakes you have. But you can paste the reports (last time and this time) here.