sheldonucr/ucr-eecs168-lab #79: [Lab04] Error with gui_start

akiyowind, 2019-03-10:
Hi, 

I'm having a strange error while trying to run gui_start. Here is a screenshot of the error:

[image]

I don't know if it matters, but I'm initializing dc_shell in the lab4-rt1 directory. I tried initializing it in the other directories as well, but received the same error.

psythefirst, 2019-03-10:
If you are using Windows, you didn't open putty with x11 set up or did not open xming.
If you are using macos, you didn't do ssh -X

akiyowind, 2019-03-10:
It was the xming. Thank you

akiyowind, 2019-03-10:
Hi,

My final layout in the Figure 49 differs slightly from the one in the lab tutorial. Just want to make sure this is ok:

[image]

psythefirst, 2019-03-10:
No problem

akiyowind, 2019-03-10:
Hi,

I am receiving this error when trying to generate clock tree:

[image]

I've tried redoing the ICC portion of the lab several times now, but keep running into the same error. Do you suggest that I just redo from the beginning @ the dc_shell porting for GCD?

psythefirst, 2019-03-10:
Maybe you did not load the sdc file. I would suggest redoing from beginning.

akiyowind, 2019-03-11:
Figured it out. I messed up in Figure 35 by not switching "Create Port" from "None" to "Top." Thanks for your help! :)