sheldonucr/ucr-eecs168-lab #86: error opening cdesigner&

gawad357, 2020-01-22:
When I was trying to start lab 2 a bit early, I got this issue pop up on my putty when trying to open cdesigner& : "Error: Neither an editor nor a viewer license is available. (LICENSE-010)". What does this error mean and how do I fix it? Please let me know whenever you can.

yushuyuan, 2020-01-22:
I have already asked Tom to get the license. You don’t need to worry about
it.

gawad357, 2020-01-22:
okay so then what do I do then b/c everytime I try to run cdesigner& now it does that and I was successful in completing labs 0 and 1

clanktian, 2020-01-22:
I also have a problem when I was trying to open cdesigner&, it said:  
"PuTTY X11 proxy: unable to connect to forwarded X server: Network error: Connection refused
cdesigner: cannot connect to X server localhost:10.0"
What should I do now? Thank you

mlo004, 2020-01-22:
I also have a problem with the license.

jinweizhang001, 2020-01-22:
The license has expired at last midnight. We have contacted the engineering system administrator last week and also followed up with him yesterday. The administrator told us he is waiting on the Synopsis's reply yesterday. We are trying to push the administrator and Synopsis to work out a way. We will notify every once this issue has been solved. Sorry about the inconvenience.

jinweizhang001, 2020-01-22:
Since Synopsis is not able to respond to us before the lab, I think that we can talk about layout design knowledge, stick diagrams and some common mistakes and debug skills in today’s lab.