sheldonucr/ucr-eecs168-lab #89: VPN crashed while trying to work on lab, now cannot edit the layout file as it's read only

arvelreeves, 2020-01-29:


jinweizhang001, 2020-01-29:
Go to your layout folder and delete the [your layout name].cdslck file. This is a lock file which automatically generates when the connection cut off abnormally. Then re-open your library manager, you should be able to open your layout in editing mode.

ryanmeoni, 2020-01-30:
I have no .cdslck file in the layout folder, and am getting the same issue. 
[image]

jinweizhang001, 2020-01-30:
Did you close your library manager and reopen it?

ryanmeoni, 2020-01-30:
For some reason using the "open recent" instead of "open design" option from the "file" drop down in the console was causing this issue. Using "open design" let me actually alter my layout.