sheldonucr/ucr-eecs168-lab #90: Lab 2 Type Error

tsandytsan, 2020-01-29:
Hi, I am on the LVS testing part of lab 2 but I am getting an error and I'm not sure on what it is. There is nothing showing up on the debugger so I am showing the details of the error that I received here. My names are all capital casing on the schematic and the layout if that helps!

[image]

jinweizhang001, 2020-01-29:
Hi, The LVS kit has an issue at present time after the license get renewed. No worries, don't do LVS for now. We are working with the vendor to solve this issue as soon as possible.

jinweizhang001, 2020-01-30:
The LVS function is now back on line after we updated the runset files. When you setup the LVS (see lab2 fig. 21), remove the default option "-ndg" in the **Option** row, just leave "-sf SPICE -quiet=0"