sheldonucr/ucr-eecs168-lab #105: Lab3 hint typo?

ryanmeoni, 2020-02-09:
In the picture below, the description under the picture says we need 2 inverters for the transistor level schematic. However, the boolean logic expressions for S and COUT in the pictures only have 1 variable ever negated (the COUT) in the S expression). Is the statement saying we need 2 inverters a typo?

jinweizhang001, 2020-02-09:
I think it depends on your own design, if one inverter is sufficient then you can only use one inverter. For example, some times your 28-transistors circuit design may gives Cout' and S', in this case you need two inverters to inverse the Cout'-> Cout and S'-> S.

ryanmeoni, 2020-02-10:
That makes sense. Another question: can I have multiple input pins with the same name in the transistor schematic all referring to the same input? I don't know if we have to use only 1 input pin for the A variable, and then draw a ton of wires from that one pin to each transistor that needs it. Being able to place multiple pins with the same name would make the schematic much cleaner.