sheldonucr/ucr-eecs168-lab #107: .tgz file Error

adela025, 2020-02-12:
I am trying to download the tar file given the normal command, but i am getting this error:

gzip: stdout: Disk quota exceeded
tar: Child returned status 1
tar: Error is not recoverable: exiting now

yushuyuan, 2020-02-12:
That means you don't have enough space in your hard disk.

adela025, 2020-02-12:
> That means you don't have enough space in your hard disk.

I have 35gb left. Is that not enough?

jinweizhang001, 2020-02-12:
Every user should have 4 gb on the bender server in your own home directory. The 35gb is the available space for the server but not for individual user. Can you try to spare some space in your home folder?

adela025, 2020-02-12:
I deleted the lab 2 tgz file. This is what I have in the eecs168 folder nothing else.
[image]