sheldonucr/ucr-eecs168-lab #116: Lab 4 Fail to Compile Verilog Source Code

MengEnLu, 2020-02-26:
[image]

clanktian, 2020-03-04:
Same problem like this. T_T

jinweizhang001, 2020-03-06:
Please read the announcement sent shortly before T_T.

_Hi all,

The server software issue has been fixed, since there is only one week left before final week, we decided that you will be only working on partial of lab4 in next week lab session. The workload normally can be finished in one hour and is easier than lab3. In Lab4, the required parts are part 3 and part 4, part 2 is skipped, the rest parts (RTL synthesis for GCD) are determined as bonus points. The lab report due by next Sunday midnight. 

TA_

closed 2020-03-06 by MengEnLu