sheldonucr/ucr-eecs168-lab #120: Getting error "Could not read the following target libraries: your_library.db"

ryanmeoni, 2020-03-06:
For part 3, I get this warning up until compile: 
[image]
Then this error when trying to compile: 
[image]

yushuyuan, 2020-03-06:
You should follow the manual, the target library is this: /usr/local/synopsys/pdk/SAED90_EDK/SAED_EDK90nm_REF/references/ChipTop/ref/saed90nm_fr/LM/saed90nm_typ.db
Please READ the manual CAREFULLY.

ryanmeoni, 2020-03-06:
I did, for some reason it was appending the file path to the default file instead of selecting the new path. Works now. 
[image]

closed 2020-03-06 by ryanmeoni