sheldonucr/ucr-eecs168-lab #121: Lab this week

ahernandez25, 2020-03-10:
Is there still lab this week? I know last week the TA was saying there might not be lab this week.

jinweizhang001, 2020-03-10:
Yes, the lab this week will be quite simple.. Please refer to the latest notification for more details.  Part 1 is for self study, part 2 is ignored, you only need to work on part 3 and part 4 of lab4. The GCD part (week 2 of lab4) is calculated as 20 bonus points. Only part 3 and part 4 are required.

ilian001, 2020-03-10:
An email was just sent out about in-person instruction being suspended. Will it affect our lab?

jinweizhang001, 2020-03-10:
I see that email. Let me check with the Professor about what he thinks.

ryanmeoni, 2020-03-11:
What do we need to include in the lab 4 pdf report?