sheldonucr/ucr-eecs168-lab #128: Horizontal vs Vertical

khamm004, 2020-03-13:
Has anyone been able to fix the problem of horizontal instead of vertical pink lines?

MCoyne1234, 2020-03-13:
I think someone else had the same issue and the answer was to check figure 37, there is a check box that has "Horizontal row" and you want to make sure it is NOT checked.

yushuyuan, 2020-03-13:
Please check Fig.37, you will find the option "Horizontal row" is not checked.

khamm004, 2020-03-13:
Yeah I saw all the other posts and the solutions provided did not help. I definitely unchecked Horizontal Row.

MCoyne1234, 2020-03-14:
I went back and redid the first part of the lab 3 times. I did it with box checked, box unchecked, and by using the provided code instead of the GUI, and I get the same output every time. Then I did the entire thing over again two more times from the beginning just to make sure. 

My GCD is fine, but the 4bit adder has horizontal lines.

yushuyuan, 2020-03-14:
Have you deleted all the files in your 4bit adder directory? If not, please first remove all the files in the directory, then redo the lab following the manual carefully. The remaining files sometimes may lead to unpredictable results.