sheldonucr/ucr-eecs168-lab #129: What exactly is Elmore Delay? DE=1/2 rc n (n-1) or 1/2 rc n n(n+1)?

clanktian, 2020-03-13:
In ppt CHAP3-3, page 6, the function of RC Elmore delay is: DE = 1/2 rc n n(n+1),
but in ppt final_review and homework3 it's DE=1/2 rc n (n-1).
Which one is correct?

jinweizhang001, 2020-03-14:
I think it should be 1/2 rc n*(n+1), if all the r and c are the same, respectively. n is the number of rc.
(1+2+3+...+n). In our homework, since the R and C are all distinguished, so you want to write out the complete expression.

clanktian, 2020-03-14:
But in HW3, problem 3, all the rc are same, the solution still writes n(n-1), so the solution is wrong?

XAWalter, 2020-03-14:
[image]
This is the equation in the book on page 177 if I am looking at the right thing.

clanktian, 2020-03-14:
[image]
then what's the different between that equation to this equation? and when should we use which euqaiton?

XAWalter, 2020-03-14:
I'm not sure, haha. lemme know if you find out anything

jinweizhang001, 2020-03-14:
That could be either a typo or just different definition on the number n (or may be shifted by 1)... If you consider n to be a large number, many sections, the delay time would actually be the same. Let me double check this .

clanktian, 2020-03-14:
Ok, thank u

jinweizhang001, 2020-03-14:
Just follow the book if the final asks you to use such formula.

XAWalter, 2020-03-14:
Thank you very much