sheldonucr/ucr-eecs168-lab #131: Homework 3 question

XAWalter, 2020-03-14:
[image]
Is there a typo in this solution? I get the same solution but to the power of -7.
This is because when C and R is plugged into the delay equation R and C are both divided by 10^2.
Is there a step I am missing or am I mistaken?

Thanks

clanktian, 2020-03-14:
[image]
[image]
I don't understand "divided by 10^2" -_-, my answer is like this, I changed all unit of variables into SI units.

XAWalter, 2020-03-14:
Thank you, I forgot to divide by 100 for R_sec. Really appreciate it!!

closed 2020-03-14 by XAWalter