sheldonucr/ucr-eecs168-lab #139: Unable to login the designer

XinyuanFu025, 2021-01-16:
Hi, anyone who has the trouble like can't launch the designer?
[image]
I'm sure the there were nothing wrong at the presteps, i could launch it last week but it doesn't work now.

XinyuanFu025, 2021-01-16:
The local host i set is 0.0 but when it launch it failed.

ychen563, 2021-01-16:
Just experienced similar issue, now I can't record the video

ychen563, 2021-01-16:
I just solved the problem. Seems Xming closed for some reason, I restart Xming and it's normal now.

XinyuanFu025, 2021-01-16:
> I just solved the problem. Seems Xming closed for some reason, I restart Xming and it's normal now.

Ah you are right! The manual said "**Again, you need to run Xming first and then putty later**", i start xming first and launched it again, it work now.
[image]