sheldonucr/ucr-eecs168-lab #151: Error launching Synopsys Galaxy Custom Designer

amo0626, 2025-09-26:
When inputting command `cdesigner&` I get the following output:

[image]

AmazingChris520, 2025-09-26:
I asked WCH support about this and he said that it is not installed yet. He will get it installed by Monday

amo0626, 2025-10-02:
I still have an issue trying to open Synopsys. [image]

TRIGGERONE, 2025-10-02:
[This](https://github.com/sheldonucr/ucr-eecs168-lab/issues/151#issuecomment-3359232527) Please use ssh -X or ssh -Y -C, instead of simply ssh, then the window can be displayed via X11.

TRIGGERONE, 2025-10-04:
Use ssh -X _usrname_@bender.engr.ucr.edu to log in Bender server

AlanSR32, 2025-10-04:
[image]
[image]
[image]]

I get access denied whenever I attempt using either one of the ssh commands when logging in

AlanSR32, 2025-10-04:
[image]

Yes I'm able to login without using ssh -X or ssh -Y -C but I run into the problem that this thread is about